hneemann/Digital #1155: Reduce components through configurable inputs

roy77, 2023-05-05:
I need a D-FF with Reset and Input Enable. To create this, I use a D-AS and a multiplexer. A basic D-FF with extendable inputs (R,S,En) would be nice. This could also merge the Register,D and D-AS components into one component.

Configurable Inputs could also reduce the amount of RAM, ROM and counter components.

prlaba, 2023-05-05:
I second this.  And please allow for inverting all control inputs, to match their physical components.

hneemann, 2023-05-23:
I have already thought about it too. The problem is that then almost all existing files would no longer work. You could add new more configurable components, but then there would be even more and there would be a lot of redundancies.

@prlaba 
Most input pins can already be inverted.

roy77, 2023-05-24:
You extend a component to the base component and leave the default values so that the component behaves as before. Then all other components can be removed from the GUI.
When loading, it is checked if the component is valid, otherwise it is converted to the base component with suitable settings.
In the next version the conversion is removed and in case of unknown components it is pointed out to use the older version for conversion. Or you leave the converter in, because it is an encapsulated component.

For D,D-AS this should work. Reg has the disadvantage that the output is centered. For this /Q is not used. If you have a configuration that activates the second output, you could put the output in the middle if it is not activated, then you could have all 3 components together and you have a much more flexible component.

roy77, 2023-06-14:
I can understand that this is a high effort and therefore not implemented.
Would it then be possible to expand the existing components with additional inputs/outputs that can be activated?
Reset, Preset, Input Enable and /Q are often needed. Of course, this can be realized by additional circuitry, but this often reduces the clarity. Of course there is always the possibility to define a subcircuit, but it always bothers me that on file level two files belong to one circuit.

hneemann, 2023-06-16:
Adding some additional optional inputs to the D-AS-FF may not be that complicated if they are disabled by default. However, the VHDL/Verilog templates would become much more complex. I will think about it.