hneemann/Digital #1156: Support for additional breadboard friendly FPGAs like Teensy, IceBreaker, ArdNANO?

dullb0yj4ck, 2023-05-13:
I love this software. It works even for showing small kids (8) how basic circuits can work and teach how to connect some every day objects in their lives, like digital 7-segment displays that are still everywhere. However, with only the TinyFPGA supported as a breadboard-friendly FPGA, it's a bit difficult to take the next step and wire it up, given the TinyFPGA isn't always available.

The Basys 3 has too much stuff built in so you can't actually teach the wiring and assignment of inputs and outputs to pins as effectively.

I'm not sure about others, but I can't find other FPGAs mentioned in the documentation for sale (in Germany) and some seem to require more expensive synthesis equipment and Microchip's software which has a much higher barrier.

Anyway, long story short, do you have any plans to add support for more simple, breadboard friendly FPGAs or a particular EE PLD platform? That would be really amazingly useful.

hneemann, 2023-05-13:
The ATF 1504 works pretty well. I use this chip in my lab exercises. Not very expensive, robust, but unfortunately only available as PLCC and a special programmer is required. The old GAL16v8 and GAL22v10 also need a programmer. And all FPGA that are not supported by the IceStorm project (i.e. everything except Lattice iCE40 FPGA) need synthesis software from the respective manufacturer.

dullb0yj4ck, 2023-05-13:
Thanks for the quick reply! I'll take a look a the Icestorm project. I haven't done this since college in 2003 with xilinx and it was a lab setup.

I had just hoped that maybe some of the stuff from other open source projects like icestudio (which support IceBreaker but doesn't seem to have good logic simulation) could be integrated here.

I just wanted to make a digital clock with my daughter. The logic was easy.

I guess the lattice iCE40 is my best bet if I don't want too high overhead costs.

closed 2023-05-13 by dullb0yj4ck