hneemann/Digital #1164: Bug in the "counting probe"

mengstr, 2023-05-28:
It seems like the probes sometimes skip the first event. In this test they are set as "Count of falling edge" and most of the time the counter on Y1 misses the first count. But every now and then it is the Y0 counter that misbehaves. I've never experienced both counters having the issue in the same run.

[image]

[CounterBug.dig.zip](https://github.com/hneemann/Digital/files/11583847/CounterBug.dig.zip)

https://github.com/hneemann/Digital/assets/325326/cfe2962d-1e9a-4b14-b5bc-18d5564a2e31

closed 2023-06-14 by hneemann

hneemann, 2023-06-14:
Thanks for pointing this tricky one out!

Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

mengstr, 2023-06-14:
The prerelease seems to work fine. The counters I use for debugging in my Discrete-USB project now all shows the expected values.  Thanks.

It was very confusing when I tried to find bugs in my design that didn't exist - when it was simply the counters that displayed slightly off values sometimes. ;-)

hneemann, 2023-06-14:
> It was very confusing when I tried to find bugs in my design that didn't exist - when it was simply the counters that displayed slightly off values sometimes. 

Sorry about that! :-)

hneemann, 2023-06-15:
@mengstr 
Have you ever had problems with wrong counts due to signal changes in the stabilization phase right after starting the simulation? I'm wondering if the counters shouldn't be reset when the stabilization phase is over and the circuit has settled. At the moment the counters are not reset.

mengstr, 2023-06-15:
Not what I can recall.... But I'll keep my eyes open for such occurrences.