hneemann/Digital #1166: Feedback circuits

Chamkey, 2023-06-12:
Hi Helmut

I'm following the Nand2Tetris course and trying to build the projects in the course visually so I can see my files in action.

I have attached a picture of my PC.dig file below
[image].

I want to increment the output no matter what, that is then fed to the first 16bit Mux which then chooses between the incremented value or the non incremented value from the Register based on if Inc is on or off. The 2nd 16bit Mux chooses either the value from the first 16bit Mux or the D value based on if load is on or off. The third 16bit Mux chooses either the value from the 2nd 16 bit Mux or 0 based on if the reset is on or off, The Register then always stores the value that it's gets and always outputs it.

I do not get an error when using their own tool called the HardwareSimulator.bat
I've attached all their tools and the project files so you may test and see that it's working on their own tool but not within Digital.

I've also added a copy of all the circuits I handcrafted in Digital so you can peruse and see what's going on.

Thanks for your help in advance.

I'm not sure if I'm getting the error because the Register is always on or if because I'm in feedback loop.
[nand2tetris.zip](https://github.com/hneemann/Digital/files/11723421/nand2tetris.zip)
[Digital Circuits.zip](https://github.com/hneemann/Digital/files/11723436/Digital.Circuits.zip)

mengstr, 2023-06-12:
Your DFF is not a `Flip-Flop` but rather just a `Gated Latch`.  When the clock input to it is high it transparently mimics the input to the output so a loop occurs.  

I.E as soon as the output changes this goes back to the multiplexers and ends up as a new value at the input again, and since the DFF is still open/transparent that new value will just go though the DFF and end up as a new value at the aouput again. And it loops and loops.

The replies here quite clearly explains the differences and inner workings of a Latch versus a Flip-Flop:
https://electronics.stackexchange.com/questions/21887/difference-between-latch-and-flip-flop

Chamkey, 2023-06-13:
@mengstr
[image]
Above is my current DFF
[image]
And above this is a D Flip Flop from a link that I came across.
I don't see a difference between the two apart from Q and Q-Bar. I only have Q as an output.
The only reason why I need to make a DFF is because Nand2Tetris doesn't actually give us a DFF it's implemented in code

hneemann, 2023-06-13:
Set the clock to high, and keep it high. Then you will see that every change at the input pin is immediately visible at the output. This is what causes the oscillation.

Change the component as follows, and the level controlled FF becomes an edge controlled FF. The oscillation will then disappear.
[image]

BTW: It makes no sense to represent an inverter by a NAND, because the NAND needs twice as many transistors as an inverter (4 instead of only 2).

Chamkey, 2023-06-14:
@hneemann 
With regards to the inverter by NAND, it's understandable that you'd use less transistors when making an inverter directly but as you can see, the course is called nand2tetris. The smallest building block that is used in the course is the NAND gate and since I'm following that course I'm literally trying to build the same computer (aka the HACK computer).
The only problem with the course is that the DFF is never given to us in NAND format and just a builtin chip (as it is with the NAND gate) I had to guess as to how it was made by using a truth table.

closed 2023-06-14 by hneemann