hneemann/Digital #1171: Export to Verilog

pcornier, 2023-06-20:
Sorry if this has already been asked before.

I have added some external Verilog modules using the 'external file' component, and wired everything like a top file. Now, I would like to export the entire design to Verilog without including the external modules. Is it possible to keep them separated from the generated file?

hneemann, 2023-06-21:
No. Until now this is not possible. The HDL export always creates a single file that contains the complete design.
The exported file is not intended to be modified manually after it has been created. It is only intended to be brought to an FPGA with an corresponding tool chain.

pcornier, 2023-06-21:
Is it something you want to implement at some point? It could be a checkbox in the external file options "skip in HDL export" as you did in "circuit specific settings". It would be an interesting option for complex designs so we can keep all the module files in their respective locations.

pcornier, 2023-06-24:
I realized that I can simply use a .dig wrapper and import it instead, with the skip option checked. Closing this now. Thanks a lot for Digital, this tool is awesome!

closed 2023-06-24 by pcornier