hneemann/Digital #1173: Cannot analyze Node External

apws, 2023-06-26:
Hi, I am trying simple verilog BLINK circuit with clk divider having clock and output. Simulation works, measurement graph too.
But I am unable to Analyse the circuit because of this "Cannot analyze Node External", so I cant generate jedec from this?

Thanks, Petr

apws, 2023-06-27:
btw, isnt this that very hard part of the thing? missing in Digital?

hneemann, 2023-06-27:
This simulator is not suitable nor intended for VHDL or Verilog analysis. If this is needed, however, an appropriate analysis/synthesis tool such as Xilinx ISE, Quartus IDE, IVerilog or GHDL can be used.

apws, 2023-06-27:
Thats okay, I am learning and it very much helps to understand lot of things, thank you.

closed 2023-06-27 by hneemann