hneemann/Digital #1178: Ram with multiple outputs

ThanasisVlioras, 2023-07-17:
Hello! I am making a processor. It has 8 data bit ram. I want to be able to read 3 bytes from it each clock cycle, each byte from a different address. 

The closest I have come to accomplishing this, is by using the provided dual port ram. It has only 2 output ports though, so 2 bytes. I thought of the gimmicky solution of tying the same inputs to 2 ram modules. They effectively behave as one, as a write on one means a write on the other etc.

Though this causes another problem. I marked both dual rams as program memory. I want them both to get the same program loaded, but it seems that the binary file is split between them. I get why program memory is implemented this way, however this is wrong behavior for my use case.

I am pretty sure I am missing something and are using Digital wrong. Help would be appreciated!

hneemann, 2023-07-18:
There are several possibilities. The easiest would be to split the address space and use a ROM for the program. Alternatively, there is a RAM in the examples for components implemented in java where you can set the number of ports, so that three are also possible (see [here](https://github.com/hneemann/digitalCustomComponents)).
Alternatively one can consider also again whether RAM with three ports must be really used. To me this seems unconventional.

ThanasisVlioras, 2023-07-18:
Do "real" cpus just take 3 clock cycles to read 3 bytes? I just did not really want to deal with multicycle logic, so I read a 16 bit instruction and potentially another byte at the same time.

About the second problem though, I believe it would be really convenient if we could just specify a file for each memory module. Don't know if it would be a priority, but it would be nice.

hneemann, 2023-07-18:
> Do "real" cpus just take 3 clock cycles to read 3 bytes?

They usually do this when data from different addresses is required. Alternatively, a 32-bit RAM can also be used. But then no unaligned accesses on byte level are possible. Or you can use four 8-bit RAMs and interconnect them with an appropriate circuit to a 32-bit memory. But in practice this is not often done. Mostly the processor will need several clocks when unaligned access is required, and the compiler then takes care that this does not happen.
But there is an example for such a circuit under Components->Library->RAMs->RAM32Bit. 

> About the second problem though, I believe it would be really convenient if we could just specify a file for each memory module. Don't know if it would be a priority, but it would be nice.

This can be done with a ROM or EEPROM. With a RAM usually no defined content is present at power-on.

ThanasisVlioras, 2023-07-18:
> Alternatively, a 32-bit RAM can also be used. But then no unaligned accesses on byte level are possible

That is really cool to think about! Though I ended up using the java component, and it works perfectly. Thank you!