hneemann/Digital #1188: Oscillation in delay at EEPROM - bug?

mengstr, 2023-08-24:
This design bugs out with an oscillation when I run it. It seems like I can't hook up a `Delay` directly to the data of an EEPROM. An `Inverter` or a `Driver` between the EEPROM and the Delay makes the problem go away.

[image]
[delaybug.zip](https://github.com/hneemann/Digital/files/12432996/delaybug.zip)

hneemann, 2023-08-26:
That is an interesting one! Thank you for pointing this out. Although I'm not sure if this is a bug or not.
The problem can also be produced in a simpler way:

[image]

This way, an infinite stream of random numbers is fed into the delay gate, which thus never reaches a steady state. 
It only occurs when the delay time is set to a value greater than one.
I have to think a bit about this problem before I understand what is best to do.

mengstr, 2023-08-26:
Yes, I see now that I can put a 1-gate delay infront of the one with the longer delay and  it works.  For the time being that is less confusing than putting a double inverter or a driver there. :)