hneemann/Digital #1190: Chaging the size of a custome components

wrigd, 2023-08-27:
Hi, Is there a way to resize the box of my custom components that was created in "New embedded Circuit" that was loaded in the main design circuit ? All of my descriptions are overlaid.

I have attached an image. 
[image]

If a compare this to the provided document, there must be a way but I have not been able to find it. Please find the image capture.
[image]

mengstr, 2023-08-27:
It took me a while to figure out how to do that the first time as well ;)

There's a `Circuit Specific Settings` in the `Edit` menu. There you can set the width and color among other things.

[image]
[image]

wrigd, 2023-08-28:
A big thank you. On the other hand, I configured the Shape for DIP-Chip in the Advanced tab just to play around the settings and I am no longer able to reset the shape to Default! I assume it's a bug.

Thanks again !