hneemann/Digital #1193: 7474 entity Verilog export

lynchaj, 2023-09-02:
Hi
I am not sure this will be considered a "bug" or not but thought you might want to know regardless.

I've made a circuit model in Digital using a mixture of logic gates and 74xx series chips from the library.

When I export the model to Verilog and load it into Quartus, all the other 74xx entities import fine *except* the 7474 entity.

I think the problem is that the 7474 entity labels its ports/pins as 1~Q, 1~RD, 1~SD, 2~RD, 2~SD, etc. and Quartus chokes on names because the tilde is not the first character.

If I manually edit the Verilog file and replace the above ports/pins to ~1Q, ~1RD, ~1SD, ~2RD, ~2SD, etc. then Quartus compiles fine.

So I think there might be a "bug" in the 7474 entity library because all of the other 74xx chips I am using compile fine in Quartus.

Maybe this is a Digital bug or maybe not.  I hope this helps regardless.  Thank you for making this great free software!

Andrew Lynch

PS, this seems like it would be an easy fix to just update the 7474 entity library entry?  Or is there more to it than that?

lynchaj, 2023-09-04:
Hi,
I've devised a workaround procedure so now the Digital Verilog export compiles/synthesizes cleanly without manual edits.

I replaced the 7474 entity in the Digital schematic with it's "included circuit" or logical expression directly.  Then removed all the input and output nodes and replaced them with direct connections in my circuit.  No more names to confuse Quartus and it produces the same output as far as I know.  At least the pinout is the same and the compiler messages/warnings seem similar except for now there is no more 7474 entity.

I hope this helps, Thanks, Andrew Lynch

hneemann, 2023-09-04:
If you change the names in the 7474.dig so that using it in the Verilog export does not cause problems anymore, you can attach the modified file here or send a pull request.

lynchaj, 2023-09-04:
Hi
Here is my modified 7474.dig file.  I hope this helps!

Thank you for offering such great free educational software!  Much appreciated!
[7474.zip](https://github.com/hneemann/Digital/files/12515392/7474.zip)

Andrew Lynch

closed 2023-09-04 by lynchaj