hneemann/Digital #1199: Allow Data Graphs to stay open after a sim ends.

taspargo, 2023-09-19:
First off, I'd like to thank you for taking the time to (re)write such a fun, useful, and complex program. I've been a little obsessed with it since I found it a couple weeks ago- I've been putting the digital circuitry from an 80's arcade game into Digital so I can simulate it and then possibly synthesize it in Verilog (being able to export in Verilog is such a great feature!). 

Anyway, my suggestion is to allow the Data Graph to stay up after a sim has been stopped (for whatever reason).  I think it would be very useful to inspect the waveforms once the sim has finished. True, you can step the clocks and check for proper functioning a step at a time, but often you want to look back at a waveform from a few steps ago. So you have to stop the sim and rerun it up to the correct time step, remember what it is you wanted to check, and then continue.

Thanks,
Tom Spargo

mengstr, 2023-09-19:
+1 for that. Having info/data left on the screen after a run has ended by error, or by pressing the stop button, would help a lot in many cases.

hneemann, 2023-09-21:
It is possible to open a data graph in the circuit specific settings. 
This graph is actually intended for debugging. 
The graph that is embedded in the circuit was originally intended for lecture demonstrations only.

taspargo, 2023-09-22:
You sure can, very nice. And the graph is zoomable too!

closed 2023-09-22 by taspargo