hneemann/Digital #1201: driver not working as documented - high-Z input is output as 1

gregretkowski, 2023-09-23:
In the docs for the driver "If the sel input is high, the output is set to the input value." however I have found that if sel is high and input is high-Z the driver outputs 1 instead of high-Z

This can be tested by setting up a simple driver circuit and running the following test set:

```
in sel out
# all as expected
1 0 Z
0 0 Z
1 1 1
0 1 0
# expected: Z, actual: 1 - this test will fail.
Z 1 Z

```

xotmatrix, 2023-09-24:
6.11 "If an input of a logical gate is set to high-Z, the read value is undefined."

[image]

prlaba, 2023-09-24:
This makes it impossible to use two Controlled Buffer components to create a bidirectional controlled buffer.  I know Digital provides its own bidirectional controlled buffer, but still…

I think the Controlled Buffer component should pass a Z through when enabled.

Paul

hneemann, 2023-09-24:
A circuit like this works perfectly fine:

[image]

gregretkowski, 2023-09-25:
I'm not really an electronics guy, I've been using Digital to learn electronics by designing an 8 bit cpu.. So the driver may be behaving like it does in the real world -- I got tripped up because the docs say the output should === the input. I put 'as documented' in the ticket because it could be working as it should IRL but just the documentation isnt clear about a high-z input resulting in output being high/low at random.

I attached a simplification of what I was running into - but have since improved my design to not use sequential drivers.

The doc quote in the ticket is when you place a driver, right-click, then click the 'help' button.

[image]

hneemann, 2023-09-25:
When a circuit is built with C-MOS logic, it is dangerous to leave an input open (high-z). This can lead to excessive current consumption in the input stage, which in extreme cases can destroy the chip. If you do this anyway, the read value is essentially random, can change over time and is very susceptible to interference of all kinds.
Therefore one avoids to leave a digital input pin unconnected or to connect it with High-Z, which is basically the same.
The simulator mimics this behavior by reading random values when a pin is unconnected.

xotmatrix, 2023-09-28:
This affects all logic gates, so maybe there should be a new section about how High-Z works in Digital. Adding a note about High-Z to every logic gate seems a little clunky.