hneemann/Digital #1210: How to implement the multiplexer...

Barbarossa1974, 2023-10-04:
The multiplexer component only has one selector location no matter how many selector bits I choose in the options box. I need four selector locations. It's unclear how the Digital multiplexer works. Can you show an example and give an explanation of how to connect the multiplexer in Digital? For instance, let's say that my selector value is 1 0 0 1, and when that is the case, I want to send input 3 to the output. Thank you for your assistance.

closed 2023-10-04 by hneemann

Barbarossa1974, 2023-10-05:
Now, I've got the same troubles with trying to hook up a register that doesn't have four connections for current state and four more for next state. I don't see how the splitter is supposed to work. An example of how to hook up four in and four out signals to the register would be helpful. Thank you for your assistance.

hneemann, 2023-10-05:
Have you read the help text in the settings dialog of the splitter?

In addition, the splitters are only required in special cases. You can connect the 4 bits from the register with a single wire to the 4 input bits of a multiplexer. It is not necessary to split the 4 bits into single wires only to join them again later.

Barbarossa1974, 2023-10-05:
Yes, I read all I could find, but being unfamiliar with the program and the lingo used in it, I didn't understand what it was saying or referring to. Perhaps I'm just thick-headed too. The help text refers to various configurations being possible, but without examples, it was difficult for me to follow what to actually do. If I were to just join the four signals into one wire and connect them, I'm not sure how I would control which number was assigned to which signal. I looked for videos on YouTube, but "Digital" is such a generic name that nothing useful came up.

Anyway, the good news is that I may have figured out the splitter enough to make it work by just experimenting. Unfortunately, I get an error when I try to run the simulation and test it out; so I don't really know. It's a circuit derived from a finite state machine. I think the problem is that the inputs need an initial state. I tried adding the generic initialization component, but, again, the help text and documentation are unfamiliar as well as the lingo that's used; so I don't know if I'm using it properly or if it's even the right thing to use. I've attached my file if you want to try running it and view the error message.

Well, I tried to attach the DIG file, but very surprisingly, the Digital file format is not supported. Okay, I saw that ZIP was supported; so, I "zipped" and attached that file. Hopefully, that works.

[FSM2.zip](https://github.com/hneemann/Digital/files/12813349/FSM2.zip)

Thanks, again, for the help.

hneemann, 2023-10-05:
The splitter in this simulator has a direction.
In your circuit, the splitter had the wrong direction (inputs and outputs were reversed). 
And then there was a short circuit in the combinational logic to generate the Ni

[FSM2.zip](https://github.com/hneemann/Digital/files/12820041/FSM2.zip)

Barbarossa1974, 2023-10-06:
Awesome! Thank you so much.

I just had to switch the order of the N inputs on the other side and fix one of the AND gate inputs that was incorrect, and now it's working perfectly. I've attached the working file. Now, I'm trying to make improvements on it. I still haven't figured out the way Digital implements a multiplexer. I've attached an image of what I'm trying to do. I did it in Digital, but I get an error message. I'm trying to replace the 5 4-input AND gate assemblage with a multiplexer.

Another tweak I wanted to do was to have a button that only sends a signal one time when it is pressed even if it is held down. Right now it continues to send a signal as long as its held down; so, I have to have the clock pretty slow. If I could configure a way to only read one time regardless of if the button is held down (it would reset when released), I could set the clock faster and get a faster response time from the circuit.

[Working FSM2.zip](https://github.com/hneemann/Digital/files/12827175/Working.FSM2.zip)
[image]
[Circuit with Multiplexer.zip](https://github.com/hneemann/Digital/files/12827177/Circuit.with.Multiplexer.zip)