hneemann/Digital #1214: Two suggestions for enhancements

mikamyara, 2023-10-09:
Hello,
  beautiful software ! I will use it with my students this year !
  I have two basic suggestions :
     - would be nice if there would be support for HiDpi screens (menus are really hard to read)
     - might be nice to have th software in a Flatpak for Linux (would help users to find it)
     
     Thanks a lot for this beautiful software !

hneemann, 2023-10-10:
It is possible to increase the font size in the settings.

closed 2024-04-20 by hneemann

egrath, 2024-09-13:
Regarding the flatpak request. I regularily build flatpaks of digital in my repository (https://github.com/egrath/Flatpaks/tree/master/digital) and are right now in the process of publishing these builds to the official flathub repository for future inclusion there, so if everything goes well, they should appear soon in the official repositories of the distribution of your choice.