hneemann/Digital #1226: Nothing connected to input 'In_1' at component 'And'

clamintus, 2023-10-27:
Hi, I'm trying to implement a finite-state machine, but sadly I'm not able to simulate my circuit.
[image]
Every time I try to start the simulation, one of the AND components complains it has nothing connected to it, even though all the inputs are clearly connected.
If I try removing the component and then placing it there again, the error just moves on to other AND components I have in the circuit.
I'm not able to find out what's wrong in my circuit, everything seems OK to me.

clamintus, 2023-10-27:
Here is my project file in case you need it: [Reti_sequenziali 1.tar.gz](https://github.com/hneemann/Digital/files/13193807/Reti_sequenziali.1.tar.gz)

hneemann, 2023-10-28:
A wire must end at a pin to make a connection. 
There is another problem in the circuit: there must be only one active output connected to a wire. If two outputs are connected to one wire, what happens when one goes high and the other goes low? In a real circuit, one of the outputs would be destroyed by an overcurrent.

I have corrected the wires that are not connected: [Reti_sequenziali 1.dig.zip](https://github.com/hneemann/Digital/files/13194498/Reti_sequenziali.1.dig.zip)

closed 2024-04-20 by hneemann