hneemann/Digital #1234: Exception error in Undo

mengstr, 2023-11-18:
It has happened twice before in this particular design. The undo stops working but everything else is fine. After restarting Digital the undo is back to working normally with no messages during startup except stating that a port has been opened. So it seems like the file can be loaded fine and there are no errors in it.
```
mats@Matss-MacBook-Pro ~ % digital
[AWT-EventQueue-0] INFO de.neemann.digital.gui.Main - open remote port 41114
```

Here's an abridged version of what's printed in the console when I try to "undo" when the problem is in effect.. Full log is attached.

```
java.lang.RuntimeException: internal error in undo
	at de.neemann.digital.gui.components.CircuitComponent.undo(CircuitComponent.java:554)
	at de.neemann.digital.gui.components.CircuitComponent.access$100(CircuitComponent.java:58)
	at de.neemann.digital.gui.components.CircuitComponent$3.actionPerformed(CircuitComponent.java:190)
	at java.desktop/javax.swing.SwingUtilities.notifyAction(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBinding(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyBindings(Unknown Source)
	at java.desktop/javax.swing.JComponent.processKeyEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(Unknown Source)
	at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
Caused by: de.neemann.digital.undo.ModifyException: Exception during event processing
 Selection deleted.
 Wire inserted.
 Component 'Text' inserted.
 Attributes of component 'Text' modified.
 Selection moved.

.....<SNIP>.....

 Component 'Pull-Up Resistor' inserted.
 Component 'Pull-Down Resistor' inserted.
 Component 'Pull-Down Resistor' inserted.
 Selection deleted.
 Component 'NOr' moved or rotated.
>
	at de.neemann.digital.undo.UndoManager.createTrace(UndoManager.java:140)
	at de.neemann.digital.undo.UndoManager.undo(UndoManager.java:121)
	at de.neemann.digital.gui.components.CircuitComponent.undo(CircuitComponent.java:552)
	... 37 more
Caused by: de.neemann.digital.undo.ModifyException: internal error: Wire not found!
	at de.neemann.digital.gui.components.modification.ModificationOfWire.getWire(ModificationOfWire.java:53)
	at de.neemann.digital.gui.components.modification.ModifyDeleteWire.modify(ModifyDeleteWire.java:29)
	at de.neemann.digital.gui.components.modification.ModifyDeleteWire.modify(ModifyDeleteWire.java:16)
	at de.neemann.digital.undo.UndoManager.undo(UndoManager.java:114)
	... 38 more
[AWT-EventQueue-0] INFO de.neemann.digital.gui.DigitalUncaughtExceptionHandler - uncaught exception log written to /Users/mats/Digital_2023-11-18_12-47-22.log

```

[Digital_2023-11-18_12-47-22.log](https://github.com/hneemann/Digital/files/13400185/Digital_2023-11-18_12-47-22.log)

hneemann, 2023-11-18:
Are you able to reproduce this error?
I have known about this bug for a long time and have tried to fix it several times without success, as it cannot be reproduced.

mengstr, 2023-11-18:
Unfortunately not. After restart it works for many hours until it just suddenly bugs out. (Which happened twice on this design)

I can't recall it ever happened in another design, but my PDP8 project that I worked on a lot a couple of years ago tended to get slower and slower doing the undos until it took several seconds per undo. But I think you fixed that issue by keeping fewer steps in the undo-buffer.

This bug is not very important considering the program doesn't crash or makes you lose any work. Digital is really really crash resilient compared to other softwares in the same size and complexity. It's nice to know that it just works and you don't need to save your design every 5 minutes.

hneemann, 2023-11-19:
See #919 and #364