hneemann/Digital #1235: Connecting Component to External IDE

jameshughes89, 2023-11-22:
From the documentation, I see that it is possible to connect an external IDE to an individual RAM component with the option "program memory" 

> Program Memory
    Makes this ROM to program memory. So it can be accessed by an external IDE.

[image]

However, I could not find details on _how_ this is done exactly within the documentation.

jameshughes89, 2023-11-22:
Despite spending about an hour looking into this before posting the issue, I seem to have found how this is done. By toggling the component (See below image)

[image]

By selecting the `Edit -> Circuit specific settings` menu option, under the `Advanced` tab, there is an option to "Preload program memory at Startup." when a file selection option below. 
[image]

This was discovered within the documentation under the "Circuit specific settings" option where it says

> Preload program memory at startup.
When simulating a processor that uses a RAM device as the program memory, it is
difficult to start this processor because the RAM contents are always initialized with
zeros at the start of the simulation. This setting allows loading data into the program
memory at startup. The program memory in the simulation must be marked as such.
> 
> Program file
File which should be loaded into the program memory at the start of the simulation.

I suppose the issue is the mention of the "external IDE", where in reality it is to add a specific hex file.

hneemann, 2023-11-22:
Both are possible: There is a TCP/IP interface that an IDE can use to transfer a HEX file to the simulator. For example, there is a plugin for VisualStudioCode which allows the debugging of assembly language for the example processor. However, you can also specify a file in the settings. This is easier, as no IDE integration is required, but not ideal, as step-by-step debugging is not possible in this way.

closed 2024-04-20 by hneemann