hneemann/Digital #1240: How to load a memory (RAM/ROM) with binary content from a .txt file?

emicrisciu, 2023-12-18:
I tried different ways of loading a memory inside a processor I am working at but something is off, I don't know why the data I write inside my txt file is loaded weirdly. I want to write instructions, in a binary format, given by an assembler inside a txt file and then load my memory with this content. How should I do it, step by step?

hneemann, 2023-12-19:
The easiest way for humans is to write the values in hexadecimal into a text file. Each value is written to a memory address starting at address 0 and can therefore be more than 8 bits long. The file must begin with the text 'v2.0 raw' in the first line and have the file extension 'hex' (see beelow).
However, intel-hex files can also be loaded. All other files are loaded as plain binary data.
```
v2.0 raw
a0 
3a3
a4 
aa6
8
```

hneemann, 2023-12-21:
I suspect that there are two components that are flagged as program memory.

emicrisciu, 2023-12-21:
You are right, the instructions are now stored correctly. And one last thing, can I somehow introduce those 16-bit instructions in binary format (from a .txt) inside the memory or is this solution (hexadecimal values) the only one possible?

hneemann, 2023-12-21:
Unfortunately, only hex is possible.

emicrisciu, 2023-12-21:
Thank you very much! I also remembered I have another problem but it's not related to this issue. For some reason, when I want to export the circuit to Verilog, I get an error pop-up saying 'No

hneemann, 2023-12-21:
You must implement a division algorithm of your choice. A division is a complex operation. If the code is to run on an FPGA, you would not leave it up to the synthesizer to decide what the division will be, how many clocks it will take, what lookup tables will be used, etc.

emicrisciu, 2023-12-21:
I see, thank you for your time!

closed 2023-12-21 by emicrisciu

hneemann, 2023-12-21:
You are welcome!