hneemann/Digital #1247: Cannot open files from DropBox folder

mhsprang, 2024-01-12:
version: v0.30

Hello,

Since I updated my iMac from Catalina to Ventura, I can no longer open files from my DropBox folder. All other folders appear to open and show files but when I select the DropBox folder, nothing is shown.

I do have full read & write access to that DropBox folder, other applications can access all files and folders in there.
Could it be some security setting because Digital is a Java application?

Best regards,
Meindert

hneemann, 2024-01-12:
I don't think so. The java application simply opens a file. Nothing special is done with file names or folder names. I expect it is some kind of MacOS security feature. Similar to the one that sometimes prevents access to the desktop.

mhsprang, 2024-01-12:
I found the solution and it is indeed a permissions problem. One needs to give the Java starter permissions to open files on the system. Found this on:
https://stackoverflow.com/questions/65469536/why-does-a-jar-file-have-no-permissions-to-read-from-disk-when-started-via-doubl

The top answer is the right one. In recent macOSses, the "Jar Launcher" is called "JavaLauncher". I also found another thread about this in this same place and I added the answer there as well.

So, all is well again. Thank you for a fantastic piece of software!!

closed 2024-01-13 by hneemann