hneemann/Digital #1259: macbook m1 iverilog

wlz987, 2024-02-09:
in setting,
set the iverilog path: /opt/homebrew/bin/iverilog
but it said no install iverilog:

由于：Verilog仿真器Icarus未安装，安装iverilog（http://iverilog.icarus.com/）并重新尝试。

hneemann, 2024-02-09:
The code that searches for the iverilog binary is quite sophisticated.  I don't think this code is not working properly. I rather suspect it is a Mac-specific "security feature" that prevents execution.

threonyl, 2024-06-20:
The issue is solved when you point Digital to the actual executable. I think something creates a problem with the symlink. Using `brew info icarus-verilog` you can locate the installation path. Then you can find the iverilog executable under `/bin/iverilog`. This solution was found by one of our course assistants :)

My executable was under this path:

`/opt/homebrew/Cellar/icarus-verilog/12.0/bin/iverilog`