hneemann/Digital #1260: Mac: How to assign Digital.jar to open .dig files

mhpolymath, 2024-02-11:
I am running macOS Sonoma 14.3.1. and JDK21.0.2
I am trying to make digital the default "open with" application for the circuit files I made with it. Since it's not an app, mac doesn't allow me to choose it to open the files, even when I check "all applications button".
Is there a workaround to do so? 
thank you.

dasobierajuk, 2024-03-11:
I have the same issue and I can't find any information online to fix it. Does anyone know how to solve this issue?

mengstr, 2024-03-16:
I did it a while back but never wrote it down. I've now recreated the steps and documented it here:
[https://matseng.org/post/associate_dig_files_with_digital_on_macos/](https://matseng.org/post/associate_dig_files_with_digital_on_macos/)

I hope that will be understandable enough and I got all the steps correct. If you don't get it working ping me and I'll see if I can help.

mengstr, 2024-03-16:
Thanks. I've added a more prominent note about the location of the jar file in the post, so hopefully that will take care of that issue a bit better.