hneemann/Digital #1265: Digital bootes to a blank screen

masroof-maindak, 2024-02-18:
Been using Digital for a couple of days on my computer and it's worked like a charm. Just tried to set it up on my laptop but no matter what I do, it refuses to show anything except  a blank screen.

I tried opening an existing `.dig` file, and I know it's opened because right-clicking at places shows attributes.

I think this might be a HiDpi issue because every other factor is constant between my Macbook and my computer. Arch Linux, Java via `jre-openjdk`, and I ran the **install.sh** script.

Running `Digital.sh` or manually starting the `.jar` shows no error files either.

[image]
[image]
[image]
[image]
[image]

hneemann, 2024-02-18:
The error can only occur if [Q], [CTRL+Q], [+] or [-] is pressed without the mouse pointer being in the window beforehand.
Does that make sense to you?

hneemann, 2024-02-18:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) v0.30-32-g468972d

masroof-maindak, 2024-02-18:
> Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) v0.30-32-g468972d

No luck whatsoever. Same blank screen. No log file.

hneemann, 2024-02-19:
if you run Digital from the console, is there still no output on the console?

hneemann, 2024-02-19:
To be honest, I have no idea what is happening. 
But I don't think it has anything to do with the code of the simulator. 
I suspect it is some problem in the JVM or the OS.

masroof-maindak, 2024-02-19:
> To be honest, I have no idea what is happening. 
> But I don't think it has anything to do with the code of the simulator. 
> I suspect it is some problem in the JVM or the OS.

That would be my first point of contention too but the problem is that they are the exact same between both my systems as well. Arch Linux + jre-openjdk package from the official repos.

Oh well, thank you anyway, feel free to close this issue.

hneemann, 2024-02-19:
You could start Digital with logging in debug mode: 
But I don't expect to see any helpful information. 
But you could at least try:

```
java -Dorg.slf4j.simpleLogger.defaultLogLevel=debug -jar Digital.jar myFile.dig
```

masroof-maindak, 2024-02-19:
[main] DEBUG de.neemann.digital.gui.Main - v0.30-32-g468972d
[AWT-EventQueue-0] DEBUG de.neemann.digital.draw.library.ElementLibraryFolder - found 129 files in /home/maindak/Documents/Programs/Digital/lib
[AWT-EventQueue-0] DEBUG de.neemann.digital.gui.Main - create with given file /home/maindak/Downloads/multiplier.dig
[AWT-EventQueue-0] DEBUG de.neemann.digital.draw.library.ElementLibrary - added library listener CircuitComponent, listeners: 1
[AWT-EventQueue-0] DEBUG de.neemann.digital.draw.library.ElementLibrary - added library listener InsertHistory, listeners: 2
[AWT-EventQueue-0] DEBUG de.neemann.digital.draw.library.ElementLibrary - added library listener LibrarySelector, listeners: 3
[AWT-EventQueue-0] DEBUG de.neemann.digital.gui.Main - loadFile: /home/maindak/Downloads/multiplier.dig
[AWT-EventQueue-0] DEBUG de.neemann.digital.gui.Main - set library root: /home/maindak/Downloads/multiplier.dig
[AWT-EventQueue-0] DEBUG de.neemann.digital.draw.library.ElementLibrary - rescan folder
[AWT-EventQueue-0] DEBUG de.neemann.digital.draw.library.ElementLibraryFolder - found 1 files in /home/maindak/Downloads
[AWT-EventQueue-0] DEBUG de.neemann.digital.draw.elements.Circuit - load /home/maindak/Downloads/multiplier.dig
[AWT-EventQueue-0] DEBUG de.neemann.digital.draw.elements.Circuit - load stream

Yeah, nothing of significant use I feel.

hneemann, 2024-02-19:
You are right: the simulator runs perfectly. 
For some weird reason, the screen output does not reach the screen.

masroof-maindak, 2024-03-31:
@hneemann I figured it out. Pretty sure this impacts about 0.01% of the Digital user-base but the culprit was the window manager of choice. I don't understand HOW the x-server could possibly impact a Java application's running but I've also learnt over the years to not question the xserver.

The problem child in this case is BSPWM, as Digital seems to be working just fine under i3.

closed 2024-04-20 by hneemann