hneemann/Digital #1268: Enhancement

HoLi-Chit, 2024-02-28:
I have built a Morse Code Electronic Keyer in Digital and It works "kinda" the way I want, but I need to be able to add a diode (1N4148 or other small signal) between two input lines of a couple of NAND gates.  I don't see any diodes that fit the need, but some tied to Vdd or Gnd.  Can it be done?  Add a diode to the list of components?

Thanks.

hneemann, 2024-02-28:
There are diodes, but they have a direction. This means that there is an input and an output. Therefore, there are also two diodes, each with the anode and the cathode as the input. For most applications, one of the two diodes can be used. Whether this is the case in your application, I could only say if I knew the exact circuit.

HoLi-Chit, 2024-03-03:
Attached is the circuit.  The diode cathode on the "Right" button and the
anode on the "Left" button such that pushing the "Left" button also pulls
the "Right" button low, enabling the "divide by 4" clock to make the DASH
component without the need hold BOTH buttons simultaneously.  Thanks.

HoLi-Chit, 2024-03-03:
Oops, got the polarity wrong on the diode for my circuit.  Reverse cathode
and anode in my original response.  Thanks again!

closed 2024-04-20 by hneemann