hneemann/Digital #1269: gpu doesnt work

aimnrt, 2024-03-04:
my gpu doesnt work (for the 3rd time) and idk how to fix it.
expected:
https://github.com/hneemann/Digital/assets/158768107/aa8a1bed-68f1-42eb-82e9-fe9edfe517c8

unexpected:
https://github.com/hneemann/Digital/assets/158768107/5b0d06dc-f209-4d32-adea-3122321b7dd3

i disconnected the rgb for the first one but enabled it for the last
[gpu_acard10.zip](https://github.com/hneemann/Digital/files/14482620/gpu_acard10.zip)
The zip file for the project

mengstr, 2024-03-04:
I'm confused over the purpose of the NAND gate. All it does is to turn off the ROM when both the horizontal and vertical D0 bits are high (I.E every other pixel on alternating lines). 

When the ROM is off (disconnected) the RGB signals are just outputting random values since nothing is driving them - giving the nice random coloured noise on the screen. ^__^

If you want to display a steady color while the ROM is off then you need to have pull ups or downs on the wires to keep them at a defined level when the ROM is not outputting anything.

But what is this circuit supposed to do? How many horizontal and vertical pixels are you expecting to be shown on the display? You are only having 8 bits of address from the horizontal counter. So that's 256 pixels. Then the same 256 pixels will be repeated for the rest of the line. (2.5 times)

You probably should have all 10 horizontal counter bits from the horizontal section going in to the ROM to avoid repeats.  For the vertical 9 bits is enough to uniquely identify 480 lines. (9 bits=512 combinations)

[image]

aimnrt, 2024-03-04:
> I'm confused over the purpose of the NAND gate. All it does is to turn off the ROM when both the horizontal and vertical D0 bits are high (I.E every other pixel on alternating lines).
> 
> When the ROM is off (disconnected) the RGB signals are just outputting random values since nothing is driving them - giving the nice random coloured noise on the screen. ^__^
> 
> If you want to display a steady color while the ROM is off then you need to have pull ups or downs on the wires to keep them at a defined level when the ROM is not outputting anything.
> 
> But what is this circuit supposed to do? How many horizontal and vertical pixels are you expecting to be shown on the display? You are only having 8 bits of address from the horizontal counter. So that's 256 pixels. Then the same 256 pixels will be repeated for the rest of the line. (2.5 times)
> 
> You probably should have all 10 horizontal counter bits from the horizontal section going in to the ROM to avoid repeats. For the vertical 9 bits is enough to uniquely identify 480 lines. (9 bits=512 combinations)
> 
> <img alt="Screenshot 2024-03-04 at 15 47 27" width="391" src="https://private-user-images.githubusercontent.com/325326/309797137-7050559d-97da-4c96-8497-a73ce6290ee4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1Njg0ODUsIm5iZiI6MTcwOTU2ODE4NSwicGF0aCI6Ii8zMjUzMjYvMzA5Nzk3MTM3LTcwNTA1NTlkLTk3ZGEtNGM5Ni04NDk3LWE3M2NlNjI5MGVlNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNFQxNjAzMDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMzg2Yzk5ZjI0OGVlYTVhMjc1MmNkYzUyYzcyOTE2NTdiYzNkMTQzZGRlNDExMjM2ZDZkYWZhNzU5ZDU3YWEwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Dghobb2BGF6D2-dHaOZxp0S8coTx4MvW_dKiL7bt9LE">

i fixed it, but now when i try to import hex images to the rom it becomes random pixels again

hneemann, 2024-04-20:
@aimnrt 
The processor/VGAExample contains a working pixel generator which can serve as an example.

closed 2024-04-20 by hneemann