hneemann/Digital #1274: Pushbutton influences oscillator

stollenm, 2024-04-03:
[SYSTEMCLOCK.zip](https://github.com/hneemann/Digital/files/14850927/SYSTEMCLOCK.zip)
If you check the attached file you will notice, that when MAN = 0 depressing PULSE will change the output CLK and the output OSC as well. This should not happen since the inputs of the mux do not change according to PULSE.

mengstr, 2024-04-03:
I'll take a stab at answering this...

I think you are interpreting the graph just as I one did. I'm used to oscilloscopes and logic analysers, but the graph here doesn't work that way. As soon as any monitored signal changes it will store that and increment to the next grid in the graph - it's really confusing.

The "Triggered Data Graph" behaves more what we are used to. Hook up the input of it to the CLK and it looks better.

I've added some counting probes to your design and you can see that the signals dosen't change even if you're clicking the pulse-button furiously. ;) 

[SYSTEMCLOCK2.dig.zip](https://github.com/hneemann/Digital/files/14851272/SYSTEMCLOCK2.dig.zip)

stollenm, 2024-04-03:
Hi mengstr. Thank you for the effort. You are absolutely right. Now I understand the difference between both graphs much better. Best regards

hneemann, 2024-04-11:
@mengstr is correct. Since the simulator does not know anything like time, but only changes that happen one after the other, it would make little sense to build something that adds time. For example, if you think for a moment about which button to press next, the graph would continue to run without anything happening in the simulation.
A real analyzer cannot stop because it does not know whether something interesting is happening or not. The simulator, however, "knows" exactly whether something is happening or not, and can trigger the graph accordingly.