hneemann/Digital #1276: Circuit symbol won't go to set height of 2

dur75945, 2024-04-13:
I am losing my mind. My circuit symbol here, when set to a heigh of two will actually be 3, (color was to make sure changes were being applied). Is there a button that I am missing? Also is there a way to apply changes to my daughter circuit without having to close and reopen my main circuit (for example say I added a 5th pin to MUL-16 in the picture and I wanted those changes to be applied to my symbol in my main file).

[image]
[image]

hneemann, 2024-04-14:
If the "Layout" shape type is used, the corners of the shape always lie on the grid, as there can also be pins at the top and bottom, and all pins must lie on the grid.
However, no pin can lie on a corner of the shape, as the labeling would then become unreadable as it would intersect with the border.
If the "Default" shape is used instead, there can be no pins at the top and bottom and the top and bottom edges of the shape lie between the grid. Only then can the height have the value two: One unit between the two input pins and half a unit each at the top and bottom.

dur75945, 2024-04-14:
Thank you so much. Is there a way to refresh the sub circuits without closing and reopening the main file?

hneemann, 2024-04-14:
If a sub circuit is opened via the shape in the main circuit, the main circuit is automatically updated as soon as the sub circuit is saved.

closed 2024-04-17 by dur75945