hneemann/Digital #1283: Triggered Data Graph misbehaving

mengstr, 2024-04-22:
In this small test to demonstrate the issue I have two screen recordings. 

The first shows the problem - If I start the clock by activating the "enable toggle" the scope screens clears, then another scope screen appears and hides repeatedly slowing down the run to a crawl. And after a while the graph is displayed the original window.  But it records far less than the set amount of samples. And it will not add anything more even if I disable the clock and then manually clicks it with the push button.  

This happens regardless of of the scope screen is closed at start of if it's already there from a previous run.

The second movie show me first adding a few manual clock ticks into the graph, then I start the freerunning mode.  Then the scope graph is continuously updated and it has no problems recording up to the set amount of samples.

I've added the dig-file for your convenience.  This is running the latest pre-release on a Macbook.

mats@Matss-MacBook-Pro hm-admin % java --version
openjdk 17.0.2 2022-01-18
OpenJDK Runtime Environment Temurin-17.0.2+8 (build 17.0.2+8)
OpenJDK 64-Bit Server VM Temurin-17.0.2+8 (build 17.0.2+8, mixed mode)

https://github.com/hneemann/Digital/assets/325326/f8b4682d-2e20-4321-a7ce-91c5de403e24

https://github.com/hneemann/Digital/assets/325326/d9038e24-bac3-4581-aea4-4c8c8dc3eafd

[FlashyScopeTest.dig.zip](https://github.com/hneemann/Digital/files/15065982/FlashyScopeTest.dig.zip)

jrkorman, 2025-02-10:
I could not recreate your problem on my system using your "circuit".

OS: Windows 11
OS version: 10.0
JRE: 22.0.2 
VM: 22.0.2+9-70

[
D:\dev\develop\Eclipse_WS_01\Digital>java -version
java version "22.0.2" 2024-07-16
Java(TM) SE Runtime Environment (build 22.0.2+9-70)
Java HotSpot(TM) 64-Bit Server VM (build 22.0.2+9-70, mixed mode, sharing)
]

Digital.jar
Build Git-Revision:v0.31-8-g1cb0d21
Build Time: 2025-02-09 22:32

Note - I completely rebuilt from source.

Jim Korman

mengstr, 2025-02-10:
I think this was resolved in the fix for issue #1331  which was another problem with the same component.