hneemann/Digital #1301: Missing gate inputs

solmli, 2024-07-11:
[PriorEncoder24B.pdf](https://github.com/user-attachments/files/16179181/PriorEncoder24B.pdf)
My 24-bit Priority Encoder keeps having Missing Gate Inputs Errors. It happened to one gate at a time as soon as I replaced it with the new one since I thought I might have created it with some wrong input parameters. It circled around and around to all the gates and I screen shot one along with its zip file to see if you help.  Thank you very much.
[PriorEncoder24b.zip](https://github.com/user-attachments/files/16179288/PriorEncoder24b.zip)

hneemann, 2024-07-11:
There is a difference between the data bits and the number of inputs. The number of data bits indicates how many bits are applied to an input. With a simple logical getter, these are all processed in parallel, as in C, for example, when two 16 bit int values are combined using an AND operator. In your circuit, the number of data bits should be set to one.

solmli, 2024-07-11:
Thank you very much for your prompt reply. You have made my day!

closed 2024-07-11 by solmli