hneemann/Digital #1305: Error "More than one output is active on a wire causing a short circuit"

neineineinein, 2024-07-23:
[VGA generator.zip](https://github.com/user-attachments/files/16354111/VGA.generator.zip)

i am following these schematics

https://eater.net/vga

mengstr, 2024-07-24:
If you follow the wires that gets highlighted in red the the error occurs you'll see that the `QA` outputs on the top two 74161 are connected a bit down the schematic - in about the same height as the EEPROM.  I'm quite sure that Ben didn't do it that way. ;).  

After removing that short it seems like the simulation at least runs without errors....

hneemann, 2024-07-24:
Once you have repaired the circuit, you must also load some image data into the EEPROM. 
To do this, you have to assign a label to the 28c256 and assign data to this label in the circuit-specific settings.

neineineinein, 2024-07-24:
> Once you have repaired the circuit, you must also load some image data into the EEPROM. To do this, you have to assign a label to the 28c256 and assign data to this label in the circuit-specific settings.

i already have some image data in the eeprom, but, the Red, Green and Blue wires are grey instead of green so there is nothing there, the code i am using can be found on eater.net/VGA ("finch.bin")