hneemann/Digital #1307: Bi directional I/O?

Mekelaina, 2024-07-27:
Love the program but I've found implementing a data bus like in older computers to be difficult. The 6502 for example, had 8 data lines with a pin for whether the CPU was reading or writing to those lines. And I've not found a decent solution other than making data in pins and data out pins. Which is fine for simulations but makes prototyping physical circuits frustrating.

hneemann, 2024-07-27:
Each pin is bi-directional by default. In fact, there are no uni-directional pins in this simulator. 
Making a pin an input or output only affects the shape of the pin and how you can interact with that shape.

shawty, 2024-08-26:
would it be possible to create a sample showing us how we might construct a bi-directional bus controlled by another signal such as those used for r/w signals?   I to have found trying to do this very frustrating.

hneemann, 2024-08-26:
It works just like a real bus. If open-collector drivers are used, you can connect all wires together; if TriState outputs are used, you need a control circuit that ensures that only one output is active. An input can be connected to the bus for reading.

Inside the sub-circuit it would look something like this:
[image]

hneemann, 2024-08-26:
And if an open-collector bus is to be used, it looks like this:

[image]

If you want to read, /out must be set to zero. And there must be a pull-up somewhere on the bus.

shawty, 2024-08-26:
Can we do "pull ups" in Digital?  That's an analog function right?  or would it suffice to just connect a line to a +ve terminal?

hneemann, 2024-08-26:
If you connect a wire to a +vcc terminal, you can never pull it down. Therefore the pull-up is necessary.
Whether a pull-up is connected to a wire or not is actually just a flag. If it is set, the wire is set to high when all connected outputs are in the high-z state. It is not really an analog function.

shawty, 2024-08-26:
Going with your sub circuit above (First Image) this is what I've basically been trying to do:

[image]

With a sub circuit inside the IO-Reg:

[image]

[image]

[image]

The I/O's etc on the external Circuit look like the following:

[image]

[image]

[image]

[image]

I think, @Mekelaina was potentially trying to do something similar to me.  In my efforts to model a 6522 VIA for example, the device has 16 r/w registers connected directly to a common data bus.  the register to access is controlled by 4 Address pins, and r/w is controlled by a regular R not W line, where it's High to read the internal register, and Low to Write to it.

shawty, 2024-08-26:
Everything works up to the point where I try to set the IO-Reg to be writable, and I get an output collision.

[image]

hneemann, 2024-08-26:
Remove the data input or set its default value to “high z”. 
If the input component outputs a value and the register also outputs a value, this leads to a short circuit.

[bus.zip](https://github.com/user-attachments/files/16752538/bus.zip)

shawty, 2024-08-26:
Ah I see what your doing now, ok, I'll have a play with that and see what transpires.  I did at one point even try to write this as a Java Component too :-)

For reference, this is what I've been trying to emulate in Digital (For quite some time now) :-D

http://6502.org/documents/datasheets/rockwell/rockwell_r6522_via.pdf

KubaO, 2025-01-16:
> Can we do "pull ups" in Digital? That's an analog function right? or would it suffice to just connect a line to a +ve terminal?

Yes. See Wires->Pull-Up Resistor. They set the signal high if it's not driven. There is also a pull-down resistor.