hneemann/Digital #1308: Add the ability to color code wires.

Mekelaina, 2024-07-30:
I've played with a few simulators. And a steam game called "turning complete" had a nifty feature where you could choose what color to make your wire/bus and this made it easier to tell wires apart in larger circuits. It would be really nice for that to be in digital. Even if there were just a hand full of different colors, being able to visually separate things in large or complex circuits would be helpful.

jameshughes89, 2024-11-26:
I would like to echo this request. Having the ability to colour code a data bus, control logic, clock, etc. would help a lot with visual clarity.

NavisGames, 2025-05-14:
Bump i guess? Would be pretty neat if colours for the busses would be implemented!

mengstr, 2025-05-17:
Personally I'd be happy enough with having "busses" shown either thicker or with a differing color.

GabbyByers, 2025-05-17:
the people desire this feature!! 🙏🙏🙏
I've been color coding some schematics that i build in digital in ms paint to help me wire up circuits on prototyping boards, this would be so helpful!

ReactorScram, 2026-10-26:
Color-coded wires was marked wontfix here - https://github.com/hneemann/Digital/issues/510 :(