hneemann/Digital #1310: Trying to use IVerilog throws "The system cannot find the path specified" error on Windows

pedromvnreis, 2024-08-03:
I've tried compiling IVerilog from source and using the setup from bleyer dot org, but neither work, the bleyer .exe throws a more generic "Node external" error.
My path is configured as `C:\iverilog\driver\iverilog.exe` and I've verified many times that it exists.
Is there a way to perhaps bypass the need for IVerilog all together?

hneemann, 2024-08-06:
The simulator is not able to analyze VHDL or Verilog code directly. This requires iVerilog, for example. However, the use of iVerilog is very slow, as the interface is only very rudimentary. If there is no particular reason for using Verilog, it is better not to use it.

pedromvnreis, 2024-08-06:
I understand that, but my issue is that Digital doesn't work with iVerilog, it could be an issue with iVerilog itself.
I've been translating my Verilog code to VHDL and using GHDL instead and it works great.

hneemann, 2024-08-06:
Why do you use an HDL at all?

pedromvnreis, 2024-08-06:
I have a component that takes in a binary number and lights up the corresponding segments in a 7-segment display on the Basys3, essentially turning a 4-bit number into a 7-bit one. 
I imagine it's much easier to do it in Verilog than by designing a circuit.

hneemann, 2024-08-06:
You could use a 4x7 bit ROM. Then it would only be a single component. :smile:

closed 2024-08-06 by pedromvnreis