hneemann/Digital #1318: Expanding components, D-Flip-Flop 4 bit input

frsj8112, 2024-08-17:
Hi, is it possible to expand the library so that it is possible to use a single component D-Flip-Flop with 4 bit input instead of 4 seperate D-Flip-Flop components?

hneemann, 2024-08-17:
Why not setting the number of data bits to 4?

frsj8112, 2024-08-17:
Sorry, I meant Number of Inputs.

Like the NAND, where you for example, can specify the amount of inputs from 2 to 4.

hneemann, 2024-08-17:
If four data bits are configured in the d-ff, the data input is 4-bit wide, so a bus can be connected that is 4 bits wide.

frsj8112, 2024-08-17:
But I can still just connect one input.
I would like to consilidate this:

[image]

To this:
[image]

hneemann, 2024-08-17:
Simply create a subcircuit that contains four d-ff.

hneemann, 2024-08-17:
Put this file alongside your main circuit.

[D-FF4.dig.zip](https://github.com/user-attachments/files/16645282/D-FF4.dig.zip)

closed 2024-10-06 by hneemann