hneemann/Digital #1324: Unable to Access Jarfile

NexoChan, 2024-09-08:
I installed the required JRE and I still got this, even through command prompt.
[image]

hneemann, 2024-09-08:
It looks like you have cloned the repository. In this case there is no jar file. This must first be built. 
Only the distribution, which is available as a zip file via the download button, contains a pre-built jar file.

NexoChan, 2024-09-09:
I'm sorry I'm kinda new here, how do I do that exactly? 
I downloaded the zip file vie download button, and it still says unable to access jarfile.

NexoChan, 2024-09-09:
Nevermind, I'm dumb, I forgot about the "releases" section...

Thankss

closed 2024-09-09 by NexoChan