hneemann/Digital #1326: Cant set the default output of a demultiplexer/multiplexer to high Z

angusbonney, 2024-09-12:
When I do this with a pin or something then it works fine but when I try to do It with a plexer then it says "The string Z is not a valid number (pos 1)!"

hneemann, 2024-09-12:
Only the demultiplexer has a default value and this cannot be high-z. 
This also seems to me to be an unusual use case, as constructs with demultiplexers or decoders usually enable components, and a high-z output is not suitable for this.
However, this behavior can be achieved with drivers:

[image]

angusbonney, 2024-09-12:
Well if its an unusual us case then I will look for a different way to do this. Btw i just realised i didn’t explain what i was doing. What im trying to do is I have a register file and I am trying to get the i/o ports on my registers to work so that i can multiplex the outputs of the registers to one i/o pin for the registers so I need to either multiplex the output to the pin or demultiplex the input to one of the registers i/o ports

hneemann, 2024-09-12:
The processor in the example folder has a register file that could possibly serve as an example.

angusbonney, 2024-09-12:
Thank you very much for the help. :)

closed 2024-09-12 by angusbonney