hneemann/Digital #1330: Idiomatic way to add "restart" component?

nametable, 2024-09-19:
Hello! I am looking at the possibility of having a component which restarts the the circuit. Currently the behavior that I am interested in is that observed when clicking on the "play" button when already playing. Currently this runs `createAndStartModel` which will close an existing model and reinitialize it.

I have noted that there is a `Stop` component already which does a `model.close()`, but how would a component reinitialize the model? The `createAndStartModel` function is in the GUI code, and I'm not sure if it would be safe for me to make it `public` and somehow call it from within a new `Restart` component.

@hneemann Do you have pointers for how this could be implemented? I would want to make sure it works also in headless mode. PS (I'm the same guy who asked about headless runtime a few months back :smile: )

If you provide direction and are interested in this feature, I'm happy to attempt an implementation and make a PR for it.

hneemann, 2024-09-22:
This would be difficult, as no component currently has the option of being reset. Each component is created at the start of the simulation and discarded at the end of the simulation.
To change this, each component with a state would have to be extended accordingly. In addition, such a state reset is not very simple. The testing effort would be very high. 

It is probably easier if the simulated circuit implements the logic for a reset itself.

nametable, 2024-09-22:
Right, I see what you're saying. It would be preferable if the circuit could reset itself.

My use case involves a processor however, which would require many cycles to write zeros to all of the RAM.

Functionally, for a processor circuit, I should just be able to reset the program counter, which I could do right now in the circuits logic. That would be more of a "soft" reset though.

After thinking about this more though, it may not make sense for Digital to be able to reset itself. Externally, I could automate killing Digital, and starting it again, analogous to switching off a computer's power, and then switching it back on.

hneemann, 2024-09-22:
> My use case involves a processor however, which would require many cycles to write zeros to all of the RAM.

But that doesn't happen when a real processor is reset either. As far as I know, no RAM chip has the option of resetting all its memory cells to zero via a RESET input or similar. The memory content is retained until the power is switched off or the cell is overwritten. A RESET option would make SRAM, for example, much more expensive. And what advantage would that bring?
No operating system or similar assumes that the memory is filled with zeros at startup. And why should it?

prlaba, 2024-09-22:
It's worth noting that Logisim's RAM component has a CLR pin.  Logism-Evolution's RAM component makes the pin optional by adding a "Use Clear pin" attribute that adds a CLR pin to the component and lets you zero all cells (the attribute is off by default). Digital could provide the same attribute to that component only, without impacting overall performance.

I agree this is not realistic, as no actual RAM has a CLR pin or a way to zero cells without writing zeros to them individually.  But I have found the option useful in the past for testing purposes.

Paul

closed 2024-10-06 by hneemann