hneemann/Digital #1331: Bug (I guess) in the Triggered Data Graph

mengstr, 2024-09-21:
If I start the circuit (attached) directly with the RUN button I only get  a few samples in the Data Graph (even if the probe says there was 100 clocks into the graph).  But if I "prime" the Data Graph by pressing the MANUAL button a few times and then RUN, I get all 100 samples into it. 

The number of samples when "unprimed"  sometimes varies from run to run. Sometimes it about 20 and sometimes just 10. Sometimes the first few samples are shifted out of the stored data so I just see samples 3 to 15. So I almost looks like it's a variable that's not initialised properly unless the graph is reset by a user action in the same cycle.

(Another thing - it would be nice if the graph was reset when starting the simulation, as it is now the old data is still visible until the first new sample is captured clearing the old data and starts to show the new ones.)

[TriggeredDataGraph.dig.txt](https://github.com/user-attachments/files/17083337/TriggeredDataGraph.dig.txt)

[image]

[image]

[image]

hneemann, 2024-09-21:
That's an interesting one! I'll have to think about that!
Thanks for pointing out!

hneemann, 2024-10-04:
Would you try version v0.31-2 found [here](https://infdigital.dhbw-mosbach.de/)?
Please let me know if this helps!

mengstr, 2024-10-04:
I've tried it on two of my current projects and it seems to work fine now. All triggered samples shows up, no annoying rapid opening/closing of the scope window and slowdown due to that.  🎉 

Two minor niggles remain.

1) The slightly annoying focus capture when the scope window is created necessitating the need to click on the main window to be able to continue using the button-shortcuts in my design. Unless it's a 1-line fix I can live with this. ;-)

2) The scope screen is not cleared at simulation start. Not really a problem, it just feels a bit strange.

closed 2024-10-04 by hneemann

mengstr, 2024-10-04:
Fantastic. I really appreciate that you fixed this issue. Thanks.

commit 2da1ce81 (2024-10-04): fixed #1331