hneemann/Digital #1333: 24 address bits max

angusbonney, 2024-09-22:
I'm trying to make a 32 bit CPU in digital and I would like 32 bits of address for rom and ram but the sim only goes up to 24 bits. I have found one way to add extra address bits by using the select input of the rom chips but doing this would be inefficient and time consuming to wire as the amount of rom chips you need to add subsequent chips grows exponentially so to add an extra 8 bits to the 24 bits of address to get to 32 bits you would need to wire up 256 rom chips which would be a massive pain and not worth it so is there anyway for me to increase it without editing the Java code myself? (I don't know java and don't really want to learn just for this). 
![image](https://github.com/user-attachments/assets/02aae3c4-b623-4975-8d35-8b4fca293703
This is a image of a proof of concept to add 4 extra address bits to rom chips with only 4 address bits making a total of 8 bits of accessible addresses.

hneemann, 2024-09-22:
If 24 address bits are used with a 32-bit data bus, this corresponds to a memory capacity of 64 megabytes. Shouldn't that be sufficient for your application? 
On the other hand, 32 address bits on a 32-bit data bus correspond to 16 gigabytes. If you need that once for RAM and once for ROM, that would be 32 gigabytes of memory. I can't imagine that being necessary in a processor simulation, because that's more than my current PC has available.

hneemann, 2024-09-22:
see also #953

angusbonney, 2024-09-22:
ah. ok thank you for the help.

angusbonney, 2024-09-22:
Sorry to reopen this issue but I would like to ask just what parts of the code would need to be edited to make 32 bits possible as I would like to do some testing with it so if you could point those files out it would be greatly appreciated.

angusbonney, 2024-09-28:
I would like to ask just what parts of the code would need to be edited to make 32 bits possible as I would like to do some testing with it so if you could point those files out it would be greatly appreciated.

hneemann, 2024-09-28:
See https://github.com/hneemann/Digital/blob/d5c5ab54c90a947e6593184d0559016a55e2d152/src/main/java/de/neemann/digital/core/element/Keys.java#L285

But keep in mind that setting the number of address bits to 32 requires 32 GBytes of memory per RAM component in the simulator. Make sure that your PC has sufficient main memory!

angusbonney, 2024-09-28:
thank you very much!

closed 2024-09-28 by angusbonney