hneemann/Digital #1341: Add feature to "Clock Input"

Enlightenment-zz, 2024-09-29:
For "Clock Input", add ability to automatically stop after X number of clock cycles.

mengstr, 2024-09-30:
Yes. A clock device that output 'n' number of pulses and then stops until triggered again would be nice to have. I tend to often need to manually step through a clocked design and I need to to it in steps of 8 or something like that.  Sometimes I want to get deep into the running of an application and I want to quickly get to like 24768 steps into the running and then start single stepping to debug a problem.

hneemann, 2024-09-30:
There is a break component. If the circuit contains a break component, the “Run To Break” button is enabled. This button clocks the simulation until the break signal is detected.

mengstr, 2024-09-30:
I've never really got the break component do anything that I like to do. :)

In a design like this I'd like to run 20 clocks (at clock speed or full speed, dosen't really matter). After the 20 steps I'd like to manually send clocks (or possible sometimes  do the propagation-stepping for some of the clocks).  But however I try to use I can't really do anything else than just propagate-step slowly slowly through the operations after the break is triggered.

I'm sure that has its usage cases, but doing several dozens of steps to propagate one clock edge isn't always practical, especially if you have stopped a little bit ahead and then want to manually clock the design.

[image]

hneemann, 2024-09-30:
I'm not sure I understood that. You can integrate several break components into the circuit and “fast-forward” the simulation to each of the break components. This gives you full control over the breakpoints.
What happens in the circuit shown if F7 is pressed after starting the simulation?

mengstr, 2024-09-30:
[On MacOS] The F7 doesn't do anything regardless if the simulation is stopped or has been started (I've by chance discovered that Start has Space as a shortcut key, something that is not apparent in the drop down menus).

The button for Run to Break is also always  disabled and can't be clicked. 

[image]
[image]

hneemann, 2024-10-01:
You have to disable the clock: The clock control needs to be manual. 
And only one clock component is allowed in the circuit to make it work.