hneemann/Digital #1348: There is a problem with the D trigger

feifeiluan2, 2024-10-20:
version: v0.31
There is a problem with the D trigger
Q marked as! Q;
! Q is marked with Q;

KubaO, 2024-12-15:
There is no "D trigger" anywhere in the program. There is a "D-Flip-flop" and "D-Flip-flop, asynchronous". These work correctly, there's no polarity mixup.

[image]

The **top** output is Q (non-inverted).
The **bottom** output is ~Q (inverted).

Note whether the D input has been inadvertently inverted. Right-click on the flip-flop, and see if there's anything in the list of the inverted inputs. If there is, you'll need to remove those inversions.

If the flip-flop looks like the below, then the D input is inverted.

[image]

KubaO, 2024-12-15:
This issue can be closed.

closed 2025-01-06 by hneemann