hneemann/Digital #1355: Support for step in sub circuits

jorjiang, 2024-10-29:
when run simulation, i can only view the values on the current level, there is no way i can go deeper and see the state of the lower level sub circuits (which is provided by logism), it is really hard to do debugging on the graphical interface, after all this is what digital is all about right

jorjiang, 2024-12-27:
@holstvoogd

prlaba, 2024-12-28:
This is the primary reason I can't use Digital for my complex CPU project.  Instead I'm left using Logisim-Evolution, a slower and less-featured simulator.

Please consider this as a high-priority feature request.

hneemann, 2024-12-29:
@prlaba 
Have you tried the probes? You can use them to display signals from sub-circuits, regardless of how deeply they are nested in sub-circuits. I use this for debugging, and in my opinion it works better than searching around in subcircuits, because all signals can be seen in one window/graph.

prlaba, 2024-12-29:
I haven't but I will.  My circuit hierarchy isn't all that deep, generally not more than 4 or 5 levels deep, so.

It's also important that I be able to pause the simulation, and resume it after correcting an error. Restarting the simulation is painful in some cases.  But I believe you've resolved this issue, yes?

Thanks.