hneemann/Digital #1357: Connecting a Multiplexer

WikipediasTop100, 2024-11-01:
Is this right way to set up a Multiplexer on Digital?
[image]

mengstr, 2024-11-01:
Does it work? If it doesn't it's not the right way. :-)  

One issue here is that you're not keeping busses the same "width". Look at the multiplexer. It has a 4-bit wide control input, so whatever you connect it to must also have 4 bits and now it's connected to a splitter that has (what it looks like) 8 output bits.

WikipediasTop100, 2024-11-01:
> Does it work? If it doesn't it's not the right way. :-)
>
> One issue here is that you're not keeping busses the same "width". Look at the multiplexer. It has a 4-bit wide control input, so whatever [...]

Does that mean I should get the values of the multiplexer to have 4 digits and connect the multiplexer to another multiplexer that has 8 outputs?

mengstr, 2024-11-01:
A multiplexer has three types of signals -  Inputs, Output and Control. The one that you are using is having 1 output and 16 inputs. 

To be able to select which one of the 16 inputs you need to give it a vale between 0 and 15 in the control pin. This requires 4 bits. So you need to connect a "bus" with the width of 4 bits to it.

One way of doing that is to convert four separate 1-bit signals into a 4-bit bus using a "splitter" - pretty much just as you did there. But it seems like your splitter is outputting 8 bits (it says 0-7 on it in your screenshot). Just change it to output 4 bits and it should work.
[image]