hneemann/Digital #1374: Thank You!

rj45, 2025-01-02:
Hi Prof. Neemann,

I just want to thank you for this wonderful program. I have got thousands of hours of enjoyment out of it.

I know burnout on open source is a real thing, and I totally understand if you are there. I hope my thank you goes some way to helping.

I just want to clarify that although I am making a similar program, there is no way to replace Digital. You have put in so many hours of amazing work, there's too many awesome things in Digital to ever "replace" it. For example, you have an amazing VGA monitor component which supports a wide range of different video modes. Or the circuit analysis tool that can produce GAL chips or other logic. Or the finite state machine UI that can really help to generate the logic to run a state machine. These must have taken a significant investment of time, and I want to say that I really appreciate it, and these are probably things I will not have time to implement in whatever I build. Very likely Digital will continue to be the only program I know of that has these features (and much much more).

If you happen to be motivated by financial contributions, I would gladly donate. But I understand if that's not motivating to you. But know I would if I could. And if my YouTube videos featuring your hard work ever actually make me significant money, I might become seriously insistent that I be able to compensate you in some way.

And know that all the "complaints" in the issues here on GitHub are because we actually use your software and hope it could be better in some ways. If we didn't use it we wouldn't be motivated to open an issue and ask for changes. Think of it this way: we have a beloved piece of software that just has one or two things that irk us that we wish were better. But we would never have found things that irk us if we hadn't invested serious time into using it.

In a lot of ways I am jealous. You have built an amazing piece of software beloved by thousands of people. Most of the software I have written is for companies that don't need it anymore. I spent countless hours writing that software for it to be thrown out. Through your software you have influenced, helped and taught thousands of people. Even if you archived Digital and never worked on it again, you still have a legacy. You did something extremely worthwhile.

So: Thank you!

(And to those reading this that know and enjoy writing Java code, I hope you consider contributing! It's a very worthwhile thing to spend your time on!)

PS: If you are burned out, I hope you do not read this as pressure to work on Digital more. It is not. Take all the time you need. There is no pressure because you do not owe us anything.

hneemann, 2025-01-03:
I don't suffer from burnout. But it's nice that you're worried. :smile: 

For my purposes, the simulator is complete. I still use it a lot in my lectures. It's just that I developed the simulator almost ten years ago. And as far as software development is concerned, I've moved on to other things in the meantime. Of course, I will continue to maintain the simulator, add a feature now and then, and sometimes there is still a bug to fix, but I consider the actual development to be complete.
Maintaining stability has become more important than adding new features. That's why I'm more reluctant to make complex changes.

fredcwbr, 2025-01-03:
I'll just add myself to the rj45 words as if they were my own.

FYI , it's being used this semester as the platform for the Digital Systems Computing Science Course here in IFGoiano - GO - Brasil (Campus Morrinhos).
My students are developing the SAP-{1,2,3} and the BIP-{1-5} on it., 

Great job .. .Thank you..

rj45, 2025-01-18:
I am glad you're not burned out.

I must say there is some disappointment that you consider Digital finished. But if I were in your shoes I am not sure I would make a different decision. 

And Digital does feel finished. It's very stable and very useful.

I do not have the time to maintain a fork, but I suppose if someone did have time, they could fork it and make some larger changes. I guess we will see if someone wants to do that.

Regardless, I really enjoy Digital, and I am extremely grateful you brought it into existence. And I very much appreciate the smaller changes you've been making now and then, and the bug fixes when they arise.

Anyway, feel free to close this issue, since it's not really an issue.

closed 2025-01-18 by hneemann