hneemann/Digital #1379: Sugestion to cerate a selector in Digital

jacneves, 2025-01-16:
version: v0.31

Quite recently, I had the problem of creating a circuit with a manual selector, with wires, with 1 input and a manually selected output,.
Is that possible, in a simple way with the Digital sw

Sincerely

JACNeves

KubaO, 2025-01-16:
Can't you use a switch? Switches can be toggled by clicking them when the simulation is running.

[image]

mengstr, 2025-01-16:
Like this or this then? (using a Multiplexer)

[image]

[image]

jacneves, 2025-01-16:
Look at the schematic. It helps.

[image: imagem.png]

16/01/2025 à(s) 17:49):

jacneves, 2025-01-16:
> Like this or this then? (using a Multiplexer)
> Screenshot.2025-01-16.at.18.55.09.png (view on web)
> <https://github.com/user-attachments/assets/6e9787cf-053a-4345-b00a-5ac608b0a4a0> Screenshot.2025-01-16.at.18.54.13.png
> (view on web)

I have made it with a input of 2 bits, plus a decoder + a set of AND gates,
between the signal and the selected line. But it becomes complex for some
teaching.
I will try with a MUX.

Thank you

KubaO, 2025-01-16:
Ok. So basically switch but with more positions than 2?I can extend the switch to allow an arbitrary number of positions.

16/01/2025 à(s) 17:49):

jacneves, 2025-01-17:
Hi,
And thank you for the reply.
Yes, I beleive it could be very helpfull.
I used a Mux and it worked well, but a multiple switch would add
flexibility.
Thanks

hneemann, 2025-01-18:
Another point worth mentioning: Since switches are real bidirectional connections that are established and disconnected while the simulation is running, they are very slow. The same applies to relays and transistors. Therefore, these components should not be used excessively.

jacneves, 2025-01-18:
Yes, I agree.
One point that I have noticed, because I use the sw for teaching, is that,
when manually controled by the user, the clovk introduces longer impulses,
than expected.
Can that issue  be solved?

JACN

hneemann, 2025-01-23:
@jacneves 
The graph has no running time. This would make little sense, especially if components are controlled manually. Therefore, the graph records changes. You can imagine it as if the time in the graph is stopped until the next change is made. If a clock is then running and a manual change is made at the same time, it appears as if the clock pulse is twice as long.