hneemann/Digital #1389: Any form of "API" for testing circuits?

avaccari, 2025-02-09:
I really like Digital and I'm using it when I teach my computer architecture course.

I was wondering if there is any way to test a circuit from "outside" Digital.

I would like to implement some automated testing for circuits submitted by my students so that they can get feedback before their final submission. So, each time they submit a .dig file, I run a series of test and compile the results in a report that I send back to the students. I can deal with the infrastructure that takes care of the submission and report, but I'm not sure if there is an option to perform the testing.
In the CONTRIBUTING.md file, it is mentioned that also the .dig files are tested so I was wondering if a framework is already available.

Right now, I am sharing a testing component but, for students who are new to the environment, the results are a little cryptic.

Thank you!

hneemann, 2025-02-11:
There is a CLI option “test”, which may be sufficient.

avaccari, 2025-02-13:
Perfect! Sorry I missed it!

closed 2025-02-13 by avaccari