hneemann/Digital #1402: error when using subcircuit

Tonyf60, 2025-03-07:
I have created a subcircuit with the aim of simulating a TON timer (IEC PLC-Programming, see image). Unfortunately, I cannot use the subcircuit because an error message appears when starting:

[image]

This is what the circuit looks like, which works as a subcircuit and does exactly what it is supposed to. The button is only there for testing. I don't need the input for T. But I hade to give the clock signal a name. I have already tested with an external clock signal. Same error.

[image]

Tonyf60, 2025-03-08:
Ok, the error does not appear now. But this means, that a clock-signal can not be used in an subcircuit. I have to add it from outside.

hneemann, 2025-03-08:
Yes! If the clock is used within a sub-circuit, it behaves like a normal input.

closed 2025-08-14 by hneemann