hneemann/Digital #1424: How to pan?

Atcold, 2025-04-11:
I feel stupid, but I haven't figured out how to pan the canvas

I've been trying every possible combination of Command, Option, Control, and Shift with scroll and drag on my Mac, but I'm unable to pan. All the scrolling results in zooming in and out.
I would have expected that scrolling vertically or horizontally would move the canvas and when Option is pressed, the behaviour changes to zooming in and out.

Help?

Atcold, 2025-04-11:
Ah, by accident, I clicked on a component and then tried dragging it and the whole canvas panned.
This makes no sense.

To move a component, you click on it and move the mouse (without grabbing it).
Clicking on a component and then dragging it will drag the canvas instead.

This sounds like a bug.

hneemann, 2025-04-11:
Dragging with right mouse button down pans the canvas.

Atcold, 2025-05-05:
Hmm… on macOS the right click is made with two fingers tap.
Dragging with two fingers is scrolling, which in this app is interpreted as zooming.
Then yes, if I force drag with two fingers, it pans.
Another way to pan is to drag an object, which does not drag the object but pans the canvas.

Everything feels quite odd.
Thanks for replying, though.