hneemann/Digital #1442: Feature Request: separate input/output spreading for splitters

ekmett, 2025-06-26:
Would it be possible to allow for separate Input/Output Spreading for wire splitters?

You can emulate it today by gathering everything up into one thing and then splitting with another spreader, obviously.

This is mostly useful if you want to regroup data from x2's to x3s or the like as you'll often still want the result bus to pitch match with the original if you are trying to model something vaguely physical.

ekmett, 2025-06-27:
One option that might be less invasive, assuming the concern would be the large body of existing diagrams using the current wire splitter, might be to allow position 'skipping'.

e.g. if you had 

this.'Input Splitting' = "1,1,1,1"
this.'Output Splitting' = "2,,2"

You could then manually adjust the stride by simply "not" putting an input/output range at a given position.

This would let enterprising folks do things like place the output in the middle of a bar as well.

ekmett, 2025-06-27:
Configuring the bits works great. There's a lot of flexibility there. That works great and I use it with reckless abandon! What I can't do is set the spacing of the pins on each side of the splitter differently. Consider if you wanted a _visual_ spacing of 3 on one side vs a 2 on the other, so that the two bars remain the same length even as I regroup. Today I have to use two wire splitters back to back to get there, because there is only one spreading option.

hneemann, 2025-06-27:
But if it's really just a question of spacing: Can't this be solved simply by cleverly laying the wires?

ekmett, 2025-06-27:
You _can_ always use two splitters back to back if you want to mix spacings, e.g.

One splitter that takes in say `1*10` and produces `10` with spreading `2`, and one that takes `10` and produces `2*5` with spreading 4.

This is how I handle this today.